FRC5263/2018Robot #3: Update programmer laptops and driver station laptops to 2018 code

elodotwe, 2018-01-08:
Uninstall all programs referencing FRC, FTC, National Instruments, Eclipse, and install the 2018 tools.

Programmer + driver stations get the [NI Update Suite](http://www.ni.com/download/first-robotics-software-2017/7183/en/). Directions for installation are [here](https://wpilib.screenstepslive.com/s/currentCS/m/getting_started/l/599670-installing-the-frc-update-suite-all-languages)

Programmer machines get:
- [Radio config utility](https://firstfrc.blob.core.windows.net/frc2018/Radio/FRC_Radio_Configuration_18_1_0.zip)
- [Java 8 JDK, get 64 bit](http://www.oracle.com/technetwork/java/javase/downloads/jdk8-downloads-2133151.html)
- [Eclipse Oxygen, 64 bit](http://www.eclipse.org/downloads/packages/eclipse-ide-java-developers/oxygen2)
- [Eclipse FRC plugins](http://first.wpi.edu/FRC/roborio/release/) (install the plugins as described [here](https://wpilib.screenstepslive.com/s/currentCS/m/getting_started/l/599679-installing-eclipse-c-java))

DylanGergely, 2018-01-08:
Swag P.C. has been updated with all programs found in this message. Java 8, Eclipse, and Radio.

EvanGeiss, 2018-01-08:
My personal laptop has been updated.

EvanGeiss, 2018-01-09:
We changed the computer set up in the lab. Dylan and I created a new command based robot project with last years gitignore, but we ran into problems between Github desktop and Eclipse. So, we had to create the Git repository on the command line.

elodotwe, 2018-01-20:
It appears that neither Swag PC nor Good Cop have the current driver station on them.

I cannot update either as I do not have this year's NI license key.

@isaiahkahler @EvanGeiss do you guys have this year's key? I can't find the NI packet for this year...

EvanGeiss, 2018-01-20:
[image]

EvanGeiss, 2018-01-20:
That should be it.

DylanGergely, 2018-01-22:
I have installed driver station on Good Cop

closed 2018-02-19 by elodotwe