FRC5263/2018Robot #4: Bench setup of roboRIO

elodotwe, 2018-01-08:
The roboRIO needs to be imaged with the 2018 image.

See [the guide on how to do that](https://wpilib.screenstepslive.com/s/currentCS/m/getting_started/l/144984-imaging-your-roborio). You'll know you won once the 2018 Driver Station (that you installed in #3) happily talks to the RIO and you can deploy from Eclipse.

elodotwe, 2018-01-08:
It's worth noting that the directions ask for us to image via USB. One of the RIOs we have seemed to have a damaged USB port last year--try to verify this, but if you run into trouble imaging via USB, you may need to image via Ethernet.

isaiahkahler, 2018-01-10:
done on both testing RIOs. waiting to see if we want working 2017 code or an updated RIO on winter bot.

elodotwe, 2018-01-10:
Will the 2018 driver station work with a 2017 robot? If not, this might be
a moot point, on account of us updating all the laptops with 2018 update
suite.

isaiahkahler, 2018-01-12:
Mr. Gill has declared winter robot as the new programmers robot, and they will use the old and crusty one as a hardware test bot (using it for pneumatics currently). Since we've already updated the RIO on it, we can switch the imaged and unimaged robot, or update the last one.

closed 2018-02-19 by elodotwe