FRC5263/2018Robot #6: Project Structure and Auton

EvanGeiss, 2018-01-10:
Needed for Driving/Turning:
-encoders
-ultrasonic
-gyro
-camera

Research:
-other control system/alternative to p.i.d
-command based code stuff
-field data output

Drive to:
-translate units to encoder values
-decide forward/backward 
-power motors
  -decide correction
    -?
  -decide  when to stop

Drive Until:
-translate units to encoder values
-know when to stop
  -slow down!

Placing Cubes:
-limit switch

Object Avoidance!!!

elodotwe, 2018-01-10:
I'd think the best place to start would be to read about Command driven
robot on the screensteps site. There's a whole section about how to design
the project. From there, dig into how to build each command (and indeed,
what each command should do).

DylanGergely, 2018-01-22:
Jacob told me to close this project.

closed 2018-01-22 by DylanGergely