FRC5263/2018Robot #7: ? commands (groups) ?

isaiahkahler, 2018-01-17:
uhhhh so like in 2017 setup, robot.java called other classes which contained their own control loops. Now we use command based programming with a more definite structure, and things get cloudy. 

my understanding of commands needs some work!!!
Would Teleop be a command or group of commands? 
And autonomous?
Will commands be called directly from robot.java? 
Will commands be called periodically, over and over, or just once, like an initialize function? 
Can subsystems call upon commands?

add more command questions and answers!!! pls and thank

isaiahkahler, 2018-01-19:
some answers:

Looked at a repository from another team, linked in the wiki. 
-OI file contains controller buttons and axis used in teleop
-OI also controls software buttons, for feeding input to commands (?) i think
-commands seem to be very short, only calling upon a method written inside a subsystem
-TeleOp will be a command (?)
-Autonomous will be a command

isaiahkahler, 2018-01-20:
CORRECTIONS AND MORE STUFF
-Command groups are a way to run commands sequentially or in parallel
-this means Autonomous will most likely be a command group not a singular command
-Logical decision code should be written inside a command (example: drive straight should be a command)
-Subsystems should be reserved for hardware manipulation (no logical code)
-command based programming is only run on a SINGLE THREAD

elodotwe, 2018-01-22:
Having had a good think and read through all the reference material...

- I think Command-based is a good framework for us to use. My recommendation would be to go forward with it. Apologies for flip-flopping.
- I think teleop is probably multiple commands. One command would be unique to teleop and not used in auton at all--"do tank drive based on driver inputs" (and this is set as the [default command for the drivebase subsystem, perhaps](https://wpilib.screenstepslive.com/s/currentCS/m/java/l/599742-default-commands)), but other commands like "open the claw" and "raise the elevator" (for instance) can be reused between auton and teleop--and those could be triggered automatically based on the `whenPressed()` declarations in the `OI` class. I'm making these assertions based on how [FRC3946 did it](https://github.com/frc3946/STEAMworks)--their project seems to be the platonic ideal of a command-based project--I think we should emulate them.
- I agree with your more recent assertion that subsystems don't have much logic to them. They simply provide an interface to the hardware, and actual logic goes inside a Command.
- Commands can trigger commands, from what I see. An interesting tidbit: `public class CommandGroup extends Command`--what that means is that CommandGroup is actually a Command--it's just a specific type of Command. So there's nothing special about a CommandGroup--it seems reasonable that a CommandGroup could even trigger another CommandGroup.

You've already identified the Drivetrain as a subsystem, which I agree with, and also placed the pneumatics in their own subsystem--that's good too.

Y'all have already identified basically all of the commands we need too:
- `TeleopDrive`: Read joystick inputs and put them to the wheels
- `DriveFor`: Drive straight using gyro correction (gyro is part of drivetrain subsystem too!) for a given distance in inches
- `Rotate`: Use gyro to rotate X degrees
- `DriveUntilObstacle`: Drive until the ultrasonic sensor (part of the drivetrain subsystem I think) indicates a certain distance
- `WaitFor`: Pause for X seconds--This will come in handy when we need to wait for ally to get out of the way

closed 2018-02-19 by elodotwe