FRC5263/2018Robot #17: WaitFor command

elodotwe, 2018-01-31:
This command should do nothing but sit and wait a number of seconds before completing.

My recommendation would be to use the [currentTimeMillis() function built into Java](https://docs.oracle.com/javase/7/docs/api/java/lang/System.html#currentTimeMillis()). When the command starts, you'll need to save (into a variable) the current time, and in the `execute()` function of the command you'll need to check whether the specified number of seconds have elapsed since you saved that time (`if (currentTimeMillis() - startTime >= howLongWeShouldWait` or something like that).

The constructor for this Command should take a number of milliseconds as a parameter:
`public Command(long waitTimeMsec)`

In case it's new to you, a millisecond is 1/1000 seconds, so 1000 milliseconds is 1 second for example.

I anticipate using this block in a list of commands--we'll do a `WaitFor(500)` followed by `DriveTo(12)` etc.

isaiahkahler, 2018-02-10:
I think there may be a better solution:
WPI Library comes with a 'setTimeout()' function
### in the constructor
``` java
public waitFor(double seconds){
    setTimeout(seconds); //amount in seconds
}
```
### isFinished method:
``` java
protected boolean isFinished() {
    return isTimedOut(); 	
}
```

elodotwe, 2018-02-19:
Has anyone tested the Wait command?

EvanGeiss, 2018-02-19:
Untested, but I don't see any reason why it won't work.