FRC5263/2018Robot #19: control camera servos

isaiahkahler, 2018-01-31:
2 continuous servos are attached to the camera, implementation controls needed.

elodotwe, 2018-02-01:
Continuous servos? As in, ones that continue rotating as long as power is
applied? How do you plan to control that?

isaiahkahler, 2018-02-02:
Yes continuous servos. I asked some builders to think about a design and they handed me the camera setup we now have attached on top of programmers' bot. I was thinking we could use the D-pad or something to control the camera's direction. Weirdly enough, the servos attached are not the same kind. There is a continuous rotation for the X axis (HSR-1425CR), and a 180 servo for the Y axis (HS-485HB). How it's now set up, we can replace the continuous rotation servo without too much hassle. The servo heads may also need some re-calibration, so we can do that too. In the screensteps website there is an example for positional servos, but not continuous, so we might be forced to switch it out.

isaiahkahler, 2018-02-13:
Update: 

- Switched out CR servo for 180 servo

- Added Servo objects in Vision subsystem

- Created Camera control with D-pad in driver control command

closed 2018-02-13 by isaiahkahler