FRC5263/2018Robot #20: DriveTo code review

elodotwe, 2018-02-01:
Indentation seems to be a mix of tabs and spaces--it is good practice to stick to one or the other. Probably Eclipse messed this up for you... If you select the entire file and hit `CTRL-I`, if I recall correctly, it will automatically make it all consistent.

https://github.com/FRC5263/2018Robot/blob/967afe246028b9470a5677a7914383cb81f8025c/src/org/usfirst/frc/team5263/robot/commands/DriveTo.java#L16-L23
These should all be `private`, not `public`--it is best practice to keep the scope of any given function/property as close to `private` as you can get away with. That way, you know folks outside your class aren't tampering with your internal state.

https://github.com/FRC5263/2018Robot/blob/967afe246028b9470a5677a7914383cb81f8025c/src/org/usfirst/frc/team5263/robot/commands/DriveTo.java#L36
This should be `private`, not `public`. The only way to run this command should be to run the whole command, not just call a function on it.

https://github.com/FRC5263/2018Robot/blob/967afe246028b9470a5677a7914383cb81f8025c/src/org/usfirst/frc/team5263/robot/commands/DriveTo.java#L39
Can you show your work here? How did you arrive at this equation? It might even be a better idea to define a few constants, like this:
```
private final int encoderClicksPerRevolution = 123;
private final double wheelRadiusInches = 2;
```
And then use those in your equation.

I would also recommend moving the logic of how to convert encoder pulses to distance out of the Command and into, perhaps, the DriveTrain subsystem. Reason being, the NavX outputs its displacement in inches already--so if we want to adapt this DriveTo command to use the NavX instead of the encoders, we should be able to do so easily if this class expects inches instead of clicks as an input.

https://github.com/FRC5263/2018Robot/blob/967afe246028b9470a5677a7914383cb81f8025c/src/org/usfirst/frc/team5263/robot/commands/DriveTo.java#L50
Please, let's abandon loop counters entirely. Instead use [Java's currentTimeMillis()](https://docs.oracle.com/javase/7/docs/api/java/lang/System.html#currentTimeMillis()). This function returns number of milliseconds since boot (or something like that--in any case it's a stopwatch that runs in 1/1000 second increments continuously. This way, it doesn't matter how frequently the system calls your `execute()` function--you'll be working off a real timebase instead of counting numbers of times your function is called.

https://github.com/FRC5263/2018Robot/blob/967afe246028b9470a5677a7914383cb81f8025c/src/org/usfirst/frc/team5263/robot/commands/DriveTo.java#L48
https://github.com/FRC5263/2018Robot/blob/967afe246028b9470a5677a7914383cb81f8025c/src/org/usfirst/frc/team5263/robot/commands/DriveTo.java#L57
Looks like this line is duplicated.

https://github.com/FRC5263/2018Robot/blob/967afe246028b9470a5677a7914383cb81f8025c/src/org/usfirst/frc/team5263/robot/commands/DriveTo.java#L50-L72
What is the thinking behind having time involved here? If the bot is at the desired distance, does it matter that it stays there for a certain amount of time?

isaiahkahler, 2018-02-05:
I worked with Evan last Friday to reconstruct this command.
### Fixed and Changed:

- No longer time based

- Variables and methods changed to `private`

- Encoder objects are replaced with method calls of the DriveTrain subsystem

### What's New:

- Now operates by approaching the target direction, and stops when that distance is hit

- Now pays attention to direction

- Command can be called, given distance and power from the constructor

- `TEMP:` Encoder values and target is output to SmartDashboard

### To do:

- Factor in inertia of robot, stop a little bit before encoders hit target

- Correction of driving if robot drives on a slant

### Ideas:

- The stopping distance factor (how far robot should stop moving to hit distance accurately), should be calculated with motor power into account. e.g. power 1.0 stops 6 inches before, power 0.5 stops 3 inches before.

- Can correction be determined by adding the left and right encoders? Motors go opposite directions, so adding the encoder values going in a straight line should always result in 0, right? If the sum is a positive number, add more power to the right; if the sum is negative, add power to the left, or something like that. Testing, thinking, and research needed!

elodotwe, 2018-02-07:
Experimentally, it seems the copy-pasted factor from last year is incorrect--the conversion from encoder clicks to distance seems funky.

@katiecrosthwaite, sorry for not posting this before tonight's meeting.

By my reckoning, we have two things at play here:
- Encoder clicks per revolution (360 for the encoders on the practice bot; check that the ones that are going onto the competition bot match)
- Diameter of the wheels (I don't know, but we can find out--probably a measuring tape is the best tool here

This is a units conversion problem. We have `Clicks` and need `inches`.

Given the diameter of the wheels, we can calculate the circumference--the distance the robot travels in one revolution. Diameter * π is circumference, so we know we'll get `diameter * π inches / revolution`.

Putting this all together, we can see the units will work out:
```
                         1 revolution   π * wheel_diameter inches
clicks (encoder.get()) * ------------ * ------------------------- = inches traveled
                          360 clicks         1 revolution
```

It's important to make this readable when you translate it into code. Right now we have a single line that does the conversion (possibly incorrectly)--and it has several numbers that aren't obvious why we'd have chosen them (the 1440 comes to mind). We don't want to repeat this mistake.

One might think you should write a comment describing this conversion, and that does help--but comments should be used to describe /why/, not /what/. The code itself should describe /what/.

If you choose good variable names and use constants instead of "magic numbers", you will be able to make sense of this in a year.

For example, I'd recommend saying something like:
```
final double wheelDiameterInches = 6.0;
final int encoderClicksPerRevolution = 360;

//and then where we calculate distance traveled...maybe inside the get() function for distance
//(this is a bit pseudo-code-y, you'll have to do a bit of tweaking to make this work)
inchesTraveled = encoderClicks / encoderClicksPerRevolution * pi * wheelDiameterInches;
```

The Encoder objects themselves should be made `private`--the rest of the code shouldn't care that there are or aren't encoders, it should only care that the drivetrain knows how far it has driven. I would think it's appropriate to have a `getInchesTraveled()` and a `resetInchesTraveled()` or something like that. Nobody in the project except this conversion function should be working in "clicks".

isaiahkahler, 2018-02-07:
I think that it's reasonable that the conversion code should be implemented int the DriveTrain `subsystem` instead of each drive or rotate `command`. I'll start working on that now.

isaiahkahler, 2018-02-07:
updates:
### Fixed and Changed:

- DriveTrain objects made private, with new methods to reach them

- isFinished variable set to false in `end` method, since commands themselves are not instantiated twice

- Encoders are reset from within the initialize function

### What's new:

- Added encoder-based correction that works! current equation is 
`leftCorrection = -1 * (leftEncoderInches - rightEncoderInches) / 10);
rightCorrection = 1 * (leftEncoderInches - rightEncoderInches) / 10);`

### To Do:

- Stop to anticipate inertia

- Combine gyro-based correction with encoder-based